Jason2866/platform-espressif32 #8: "HTTPUpdateServer.h" does not find this file

koltinov, 2022-01-26:
I'm trying to build an arduino example from the folder - C:\.platformio\packages\framework-arduinoespressif32\libraries\HTTPUpdateServer does
not compile with an error -

```
> Executing task: C:\.platformio\penv\Scripts\platformio.exe run <

Processing esp32dev (platform: https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.2.1patch2/platform-tasmota-espressif32-2.0.2.1.zip; framework: arduino, espidf; board: esp32dev)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
FILESYSTEM  spiffs
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.0.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.0.2+1
 - framework-espidf 3.40400.0+rc1 (4.4.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30200.211025 (3.2.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0) 
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\src\main.o
Archiving .pio\build\esp32dev\esp-idf\app_update\libapp_update.a
Archiving .pio\build\esp32dev\esp-idf\asio\libasio.a
src/main.cpp:9:10: fatal error: HTTPUpdateServer.h: No such file or directory

**************************************************************************
* Looking for HTTPUpdateServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:HTTPUpdateServer.h"
* Web  > https://platformio.org/lib/search?query=header:HTTPUpdateServer.h
*
**************************************************************************

 #include <HTTPUpdateServer.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\esp32dev\bootloader_support\src\bootloader_common.o
*** [.pio\build\esp32dev\src\main.o] Error 1
```

if I move the "HTTPUpdateServer" folder to the platformio /lib project folder, then it compiles without errors

my platformio.ini
```
[env:esp32dev]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.2.1patch2/platform-tasmota-espressif32-2.0.2.1.zip
framework = 
	arduino
	espidf
board = esp32dev
upload_speed = 921600
monitor_speed = 115200
monitor_filters = 
	time
	esp32_exception_decoder
build_flags = 
	-D VERSION_F=1
	-D CORE_DEBUG_LEVEL=3
	-D CONFIG_ARDUHAL_LOG_COLORS=1
	-D ARDUINO_ARCH_ESP32=1
	-D ESP32=1
	-D ESP_PLATFORM=1
	-D ARDUINO=10805
```

question: where are the inclusion paths spelled out during compilation so that I can fix it?

Jason2866, 2022-01-27:
Do NOT add framework `espidf` when you want to build a Arduino app.

koltinov, 2022-01-27:
you misunderstood me. this is just an example. I have a project on espidf, it includes some arduino libraries. other arduino libraries compile normally. here at the moment there is a need to add the possibility of web-ota updates. I wanted to do this with the arduino library, but it does not compile as part of the project. I tried to build a standard example separately and the error is the same.

koltinov, 2022-01-27:
Other arduino examples work in the same mode. For example HTTPUpdate.h finds and compiles, but HTTPUpdateServer.h does not find.

Jason2866, 2022-01-27:
If you use a Arduino lib with IDF you may have to change manually CMakeLists.txt
Did you run `make menuconfig` ?
Using Arduino code / libs with IDF is not always easy.
All CI build examples do work, so there is no bug in general.
Sorry i cant help you any further. espressif has a good forum for such questions

closed 2022-01-30 by koltinov