Jason2866/platform-espressif32 #11: UART IRQ broken

gemu2015, 2022-02-15:
@Jason2866  (master of the cores)
Hi ,
wanted to inform you that this PR broke Hardwareserial in some situations.
https://github.com/espressif/arduino-esp32/pull/6134

i had some users complaining about SML stopped working and investigated the issue

this is by the way a stupid modification. it creates a new task which uses 2k of RAM to monitor the UART IRQ. 
it also causes unnecessary cpu cycles.

it adds the call uartOnReceive
which may be of interest for some applications but definitely should be optional and not default. 
why not enable this feature only when requested ?

reverting this pr brings things up to normal.

Jason2866, 2022-02-15:
@gemu2015  Servus, greetings from South Germany.
Could you open a issue in Arduino Github too?
I will revert in our build. Dont like that because every difference to upstream is a pain to handle/maintain.

Jason2866, 2022-02-15:
@gemu2015  Give it a try. Place in platformio override file
```
[env:tasmota32_base]
; *** Uncomment next lines ";" to enable development Tasmota Arduino version ESP32
platform                = https://github.com/Jason2866/platform-espressif32.git#IDF44/ESP32-S3
platform_packages       = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/664/framework-arduinoespressif32-v4.4_dev-73c0f44528.tar.gz
build_unflags           = ${esp32_defaults.build_unflags}
build_flags             = ${esp32_defaults.build_flags}
```
Needs https://github.com/arendst/Tasmota/pull/14861 for fixing onewire
which i have just merged now

closed 2022-02-15 by Jason2866

SuGlider, 2022-02-17:
@Jason2866 - tht PR [#6134](https://github.com/espressif/arduino-esp32/pull/6134) must be applied with PR #[6201 ](https://github.com/espressif/arduino-esp32/pull/6201)to fix it.