Jason2866/platform-espressif32 #12: PSRAM type selection missing for S3

Jason2866, 2022-03-04:
S3 introduces a new type of PSRAM (OPTI). There is currently no option to select the FLASH PSRAM type combinations with Platformio settings. https://github.com/espressif/arduino-esp32/discussions/6197#discussioncomment-2295761

@maxgerhardt do you have a idea how to implement nicely?

maxgerhardt, 2022-03-04:
The way I see it, you have 3 different PSRAM types as folders in https://github.com/espressif/arduino-esp32/tree/esp32-s3-support/tools/sdk/esp32s3, which are 

The `boards.txt` hardcodes then this setting for each board

>esp32s3.build.flash_type=qspi
>esp32s3.build.psram_type=qspi
>esp32s3.build.memory_type={build.flash_type}_{build.psram_type}
>...
>esp32s3box.build.memory_type=qspi_opi

The `platform.txt` is then using that option in a `-L` flag

>`"-L{compiler.sdk.path}/{build.memory_type}"`

However, https://github.com/espressif/arduino-esp32/blob/esp32-s3-support/tools/platformio-build-esp32s3.py#L298 hardcodes the used library include path as 

```py
    LIBPATH=[
        join(FRAMEWORK_DIR, "tools", "sdk", "esp32s3", "lib"),
        join(FRAMEWORK_DIR, "tools", "sdk", "esp32s3", "ld"),
        join(FRAMEWORK_DIR, "tools", "sdk", "esp32s3", "qspi_qspi")
    ],
```

So I would suggest adding another field in the PlatformIO board definitions, e.g. in the `build.arduino` object which is currently

https://github.com/Jason2866/platform-espressif32/blob/e104e061389306d8eb8b4a63176ea141150b447a/boards/esp32-s3-devkitc-1.json#L1-L5

to e.g. to

```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "memory_type": "qspi_opi"
    },
```

and then adapt the build script part above accordingly

```py
    LIBPATH=[
        join(FRAMEWORK_DIR, "tools", "sdk", "esp32s3", "lib"),
        join(FRAMEWORK_DIR, "tools", "sdk", "esp32s3", "ld"),
        join(FRAMEWORK_DIR, "tools", "sdk", "esp32s3", env.BoardConfig().get("build.arduino.memory_type", "qspi_qspi"))
    ],
```

So the fallback is, if the option does not exist, regular QSPI (flash), otherwise the option as declared by build.arudino.memory_type.

Jason2866, 2022-03-04:
Thx, a lot! Looks great. @me-no-dev can we do it this way?

me-no-dev, 2022-03-05:
yes we can. There is one thing though... I have not finalized the OPI options yet and it seems that e will need more than just that define for them. PSRAM will need to be enabled on boot by IDF instead of Arduino, so besides that setting, another one to show that psram is booted will be needed.

Jason2866, 2022-03-05:
Mhh, what is speaking against always enabling PSRAM in sdkconfig with
```
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
```
We do this in our fork and it works well. We check in our project (could be done in Arduino too) if PSRAM workaround is needed and disable PSRAM when the needed workaround code is not there. By doing this for many esp32 devices PSRAM support is enabled without user has to do anything. We have no issues since we do this.

Jason2866, 2022-03-05:
We could test this approach first in our fork. The only change for this would be the change in the build script of the S3 (which is backwards compatible).

me-no-dev, 2022-03-05:
psram is enabled, but not initialized by IDF. It's the later initialization of OPI psram when flash is QSPI that is the problem. All other options we can init in Arduino on boot

Jason2866, 2022-03-11:
Closing since it is implemented now as suggested from @maxgerhardt 
Thx!

closed 2022-03-11 by Jason2866