Jason2866/platform-espressif32 #38: Error: Could not find one of 'package.json' manifest files in the package [platform-espressif32-2.0.5.1.zip]

QGB, 2022-10-09:
```
platform-espressif32-2.0.5.1.zip; board: esp32-c3-devkitm-1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/tasmota/esptool/releases/download/v4.3/esptool-4.3.zip
Unpacking  [####################################]  100%
Error: Could not find one of 'package.json' manifest files in the package
```

Jason2866, 2022-10-09:
Could not reproduce. Please try deleting folder `.cache`

QGB, 2022-10-09:
delete ``C:\Users\qgb\.platformio\.cache\downloads`  

problem solved

```
Tool Manager: Installing http://192.168.1.3/C%3A/Users/qgb/Downloads/esptool-4.3.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40300.220914 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
```

QGB, 2022-10-09:
@Jason2866 

but when I compile and run arduino esp32c3 .still got error:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf64a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a0
load:0x403ce710,len:0x216c
SHA-256 comparison failed:
Calculated: 236799dd10e5b883b047aa7266eb6dff7e484fbc6e35190f5e5bc5035caac734
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf64a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a0
load:0x403ce710,len:0x216c
SHA-256 comparison failed:
Calculated: 236799dd10e5b883b047aa7266eb6dff7e484fbc6e35190f5e5bc5035caac734
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf64a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a0
load:0x403ce710,len:0x216c
SHA-256 comparison failed:
Calculated: 236799dd10e5b883b047aa7266eb6dff7e484fbc6e35190f5e5bc5035caac734
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf64a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a0
load:0x403ce710,len:0x216c
SHA-256 comparison failed:
Calculated: 236799dd10e5b883b047aa7266eb6dff7e484fbc6e35190f5e5bc5035caac734
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403cc710
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cf64a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0xf4
load:0x403cc710,len:0x8a0
load:0x403ce710,len:0x216c
SHA-256 comparison failed:
Calculated: 236799dd10e5b883b047aa7266eb6dff7e484fbc6e35190f5e5bc5035caac734
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
```

QGB, 2022-10-09:
```
Tool Manager: Installing http://192.168.1.3/C%3A/Users/qgb/Downloads/esptool-4.3.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.40300.220914 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (2.0.5) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduino-ITEAD @ 2.0.5
 - framework-arduino-solo1 @ 2.0.5
 - framework-arduinoespressif32 @ 2.0.5
 - tool-esptoolpy @ 1.40300.220914 (4.3.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32-c3-devkitm-1\src\main.cpp.o
Generating partitions .pio\build\esp32-c3-devkitm-1\partitions.bin
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\FirmwareMSC.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\Print.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\USB.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\USBCDC.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\USBMSC.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\WString.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-i2c-slave.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-i2c.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-rgb-led.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-time.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\firmware_msc_fat.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\libb64\cencode.c.o
Compiling .pio\build\esp32-c3-devkitm-1\FrameworkArduino\main.cpp.o
[... 42 lines cut ...]
Wrote 11072 bytes (8360 compressed) at 0x00000000 in 0.3 seconds (effective 261.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 516.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 753.6 kbit/s)...
Hash of data verified.
Compressed 132544 bytes to 76621...
Writing at 0x00010000... (20 %)
Writing at 0x0001764f... (40 %)
Writing at 0x00020ec7... (60 %)
Writing at 0x00026dec... (80 %)
Writing at 0x0002c9df... (100 %)
Wrote 132544 bytes (76621 compressed) at 0x00010000 in 2.3 seconds (effective 454.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================== [SUCCESS] Took 46.99 seconds ==============================================
 *  终端将被任务重用，按任意键关闭。 

```

Jason2866, 2022-10-09:
Mhh, can be wrong flash size / partition scheme

closed 2022-10-09 by Jason2866

QGB, 2022-10-09:
@Jason2866  It is 4MB flash, I use  esp-idf to compile and run  hello_world successfully. But platformio does not.

Jason2866, 2022-10-09:
We use this build to compile Tasmota. All variants are working ESP32, ESP32-C3, ESP32-S2 and ESP32-S3. With Linux, Windows and MacOS.
You can see in the CI builds https://github.com/arendst/Tasmota/actions/runs/3214824926