Jason2866/platform-espressif32 #46: Kernel panic when try to use digest authentification

pablozg, 2023-04-26:
Hi, Using your release in order to avoid the dhcp connect issue with some routers,

```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.04.02/platform-espressif32.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/1322/framework-arduinoespressif32.zip
```

I get a collateral issue:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Decoding stack results
0x40089c4d: memchr at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/string/memchr.c line 118
0x4018cd5a: _printf_i at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 219
0x4018bffd: _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/nano-vfprintf.c line 636
0x4018ec55: _vasprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vasprintf.c line 62
0x4018ec82: vasprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vasprintf.c line 38
0x4012d462: md5_printf at /home/src/esp32-arduino-lib-builder/esp-idf/components/esp_http_client/lib/http_auth.c line 44
0x4012d5e8: http_auth_digest at /home/src/esp32-arduino-lib-builder/esp-idf/components/esp_http_client/lib/http_auth.c line 110
0x4012c45e: esp_http_client_prepare at /home/src/esp32-arduino-lib-builder/esp-idf/components/esp_http_client/esp_http_client.c line 543
0x4012cdfb: esp_http_client_connect at /home/src/esp32-arduino-lib-builder/esp-idf/components/esp_http_client/esp_http_client.c line 1247
0x4012d043: esp_http_client_open at /home/src/esp32-arduino-lib-builder/esp-idf/components/esp_http_client/esp_http_client.c line 1421
0x400db192: open_envoy_connection() at src/httpClientIDF.cpp line 217
0x400db25a: connect_envoy() at src/httpClientIDF.cpp line 231
0x400db287: http_rest_envoy() at src/httpClientIDF.cpp line 250
0x400db37d: http_request_task(void*) at src/httpClientIDF.cpp line 48
```

when return back to:

```
platform = espressif32
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.8
```

the issue  is gone.

I tried to compile my own version, but the changes I made don't work.

Do you know what option must change on menuconfig to fix it?

thanks.

Jason2866, 2023-04-28:
This is enough
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.04.02/platform-espressif32.zip
```

Jason2866, 2023-04-28:
Can you provide a small test sketch where you encounter the problem?

Jason2866, 2023-05-13:
Closing since no feedback

closed 2023-05-13 by Jason2866

pablozg, 2023-05-14:
I'm sorry, last week I was very busy, here you have a test code:

```
#include "WiFi.h"
#include <sys/param.h>
#include "esp_http_client.h"
#include "esp_tls.h"

unsigned long boot_time;

char ssid[] = "ssid";     //  your network SSID (name)
char pass[] = "pass";  // your network password

#define MAX_HTTP_RECV_BUFFER 512
#define MAX_HTTP_OUTPUT_BUFFER 2048

static char local_response_buffer[MAX_HTTP_OUTPUT_BUFFER] = {0};

esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            Serial.printf("HTTP_EVENT_ERROR\n");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            Serial.printf("HTTP_EVENT_ON_CONNECTED\n");
            break;
        case HTTP_EVENT_HEADER_SENT:
            Serial.printf("HTTP_EVENT_HEADER_SENT\n");
            break;
        case HTTP_EVENT_ON_HEADER:
            Serial.printf("HTTP_EVENT_ON_HEADER, key=%s, value=%s\n", evt->header_key, evt->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            Serial.printf("HTTP_EVENT_ON_DATA, len=%d\n", evt->data_len);
            break;
        case HTTP_EVENT_ON_FINISH:
            Serial.printf("HTTP_EVENT_ON_FINISH\n");
            break;
        case HTTP_EVENT_DISCONNECTED:
            Serial.printf("HTTP_EVENT_DISCONNECTED\n");
            break;
    }
    return ESP_OK;
}

static void http_auth_digest(void)
{
    esp_http_client_config_t config = {
        // .url = "http://user:passwd@httpbin.org/digest-auth/auth/user/passwd/MD5/never",
        .url = "http://httpbin.org/digest-auth/auth/user/passwd/MD5/never",
        .username = "user",
        .password = "passwd",
        .auth_type = HTTP_AUTH_TYPE_DIGEST,
        .timeout_ms = 3000,
        .event_handler = _http_event_handler,
        .transport_type = HTTP_TRANSPORT_OVER_TCP,
        .buffer_size = 1024,
        .buffer_size_tx = 512,
    };
    
    esp_http_client_handle_t client = esp_http_client_init(&config);

    Serial.printf("Opening non-secure connection\n");
    esp_err_t err = esp_http_client_open(client, 0);
    if (err == ESP_OK) {
        Serial.printf("HTTP Digest Auth Status = %d, content_length = %" PRIu64 "\n",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
    } else {
        Serial.printf("Error perform http request %s\n", esp_err_to_name(err));
    }
    
    Serial.printf("Reading all headers\n");
    esp_http_client_fetch_headers(client);
    Serial.printf("Adding digest auth header\n");
    esp_http_client_add_auth(client); // In this call do the kernel panic
    Serial.printf("Closing connection\n");
    esp_http_client_close(client);
    Serial.printf("Performing request again\n");
    esp_http_client_set_header(client, "Connection", "Keep-Alive");
    esp_http_client_set_header(client, "Cache-Control", "max-age=0");
    err = esp_http_client_open(client, 0);
    if (err == ESP_OK) {
        Serial.printf("HTTP Digest Auth Status = %d, content_length = %" PRIu64 "\n",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
        
        size_t readLen = esp_http_client_read(client, local_response_buffer, 512); // 128
        
        if (readLen >= 0) {
            Serial.printf("Len: %d, Read data: %s\n", readLen, local_response_buffer);
        } else { Serial.printf("Fail to read data\n"); }
    } else {
        Serial.printf("Error perform http request %s\n", esp_err_to_name(err));
    }
    Serial.printf("Cleaning up\n");
    esp_http_client_cleanup(client);
}

void setup()
{
   Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  WiFi.begin(ssid, pass);
  delay(100);

  // attempt to connect to Wifi network:

  while (WiFi.status() != WL_CONNECTED) {

    Serial.print("Attempting to connect to WPA SSID: ");

    Serial.println(ssid);

    delay(1000);

  }

   Serial.print("You're connected to the network\n");
}

void loop()
{

  http_auth_digest();

  delay(2000);
}
```

Using the default config from espressif arduino builder and your lwip timeout branch, I was able to run my code without this issue.

Platformio config:

[env:heltec_wifi_kit_32]
framework = arduino
board = heltec_wifi_kit_32

; Basado en idf 4.4.4
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.04.02/platform-espressif32.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/1322/framework-arduinoespressif32.zip

Jason2866, 2023-05-14:
Probably because we use the ROM inbuilt nano lib. Not all features are supported for printf.
Could be `serial.printf` is not supported. Thanks for the feedback.
Since we use only features which are supported in nano lib, we will keep.

pablozg, 2023-05-14:
The problem isn't the printf (I put it for clarify), in my original code no Serial.printf or log_i or ESP_LOGI are used. The kernel panic happens when call the 'esp_http_client_add_auth(client);' function.

Jason2866, 2023-05-14:
Interesting issue. After some Core panic reboots, it is working.

Jason2866, 2023-05-14:
It is a timeout issue. When i change to `.timeout_ms = 500,` it crashes seldom. Lowering more solves crahing.

Jason2866, 2023-05-14:
Wit "400" get no crash and this output
```
Attempting to connect to WPA SSID: Jason_Home_WLAN
You're connected to the network
Opening non-secure connection
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = 0, content_length = lu
Reading all headers
Adding digest auth header
Closing connection
HTTP_EVENT_DISCONNECTED
Performing request again
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = -1, content_length = lu
Len: 0, Read data: 
Cleaning up
HTTP_EVENT_DISCONNECTED
Opening non-secure connection
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = 0, content_length = lu
Reading all headers
Adding digest auth header
Closing connection
HTTP_EVENT_DISCONNECTED
Performing request again
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = -1, content_length = lu
Len: 0, Read data: 
Cleaning up
HTTP_EVENT_DISCONNECTED
Opening non-secure connection
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = 0, content_length = lu
Reading all headers
Adding digest auth header
Closing connection
HTTP_EVENT_DISCONNECTED
Performing request again
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = -1, content_length = lu
Len: 0, Read data: 
Cleaning up
HTTP_EVENT_DISCONNECTED
Opening non-secure connection
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = 0, content_length = lu
Reading all headers
Adding digest auth header
Closing connection
HTTP_EVENT_DISCONNECTED
Performing request again
HTTP_EVENT_ERROR
Error perform http request UNKNOWN ERROR
Cleaning up
HTTP_EVENT_DISCONNECTED
Opening non-secure connection
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = 0, content_length = lu
Reading all headers
Adding digest auth header
Closing connection
HTTP_EVENT_DISCONNECTED
Performing request again
HTTP_EVENT_ON_CONNECTED
HTTP_EVENT_HEADER_SENT
HTTP Digest Auth Status = -1, content_length = lu
Len: 0, Read data: 
Cleaning up
HTTP_EVENT_DISCONNECTED
Opening non-secure connection
```

pablozg, 2023-05-14:
Yes, it's a Interesting issue, using your builder with the original values from espressif and the lwip-timeout branch works well, so by the moment I will use my custom build with lwip-timeout, seems to fix some the issues with the dhcp server connection.

thanks for you work.