Jason2866/platform-espressif32 #51: File system not available for esp32 c2

HalfSweet, 2023-08-29:
I can't mount any filesystem on esp32 c2.

For example I am using SPIFFS and when I call `SPIFFS.begin()` to mount, this triggers an error
```
[ 921][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: 261␍␊
```

My code is very simple

```cpp
#include <SPIFFS.h>

void setup()
{

  Serial.begin(115200);
  Serial.println("Hello");
  delay(100);
  SPIFFS.begin();
}
```

closed 2023-09-01 by Jason2866

HalfSweet, 2023-09-01:
Of course, LittleFS triggers a similar error, which appears to be a file system mount failure.

I created a new partition table and I'm not sure if this is a problem or not

```csv
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x100000,
app1,     app,  ota_1,   0x110000, 0x100000,
spiffs,   data, spiffs,  0x210000,0x7D000,
```

Jason2866, 2023-09-01:
No problem with using LittleFS and the C2. You can [install Tasmota](https://jason2866.github.io/Tasmota-specials/) on the C2 and see it working ;-)