Jason2866/platform-espressif32 #52: Upload Problem With ESP32-H2

rvbc1, 2023-11-29:
Hello,

I've been trying out your IDF5 branch and here's my configuration in platformio.ini:

```
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
board = esp32-h2-devkitm-1
framework = arduino

```

Everything seems to be working fine so far (the code compiles), but I encounter an issue when trying to upload the firmware. Here's what I see:

```
Building in release mode
Retrieving maximum program size build/local/firmware.elf
Checking size build/local/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.2% (used 7132 bytes from 327680 bytes)
Flash: [=         ]   6.4% (used 125960 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: /dev/ttyUSB1
Uploading build/local/firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument --flash_freq/-ff: invalid choice: '64m' (choose from 'keep', '80m', '60m', '48m', '40m', '30m', '26m', '24m', '20m', '16m', '15m', '12m')
*** [upload] Error 2
```

In Arduino IDE 2.2.1 and Arduino 3.0.0-alpha2, the compilation and upload process both work fine. Could you possibly help me figure out how to fix this issue? Or could you guide me on where to adjust the settings for esptool to get this working correctly?

rvbc1, 2023-11-29:
Ok, I found a workaround.

I added the following line to platformio.ini:

I added to platformio.ini
`extra_scripts = pre:modify_upload.py`

And created a "modify_upload.py" script:

```
Import("env")

def before_upload(source, target, env):
    uploader_flags = env['UPLOADERFLAGS']
    print("Oryginalne flagi uploadera: ", uploader_flags)

    try:
        index = uploader_flags.index('--flash_freq') + 1
        uploader_flags[index] = 'keep'
    except ValueError:
        print("Not found '--flash_freq' in flags")

    env.Replace(UPLOADERFLAGS=uploader_flags)

env.AddPreAction("upload", before_upload)
```

I discovered the "keep" parameter in the Arduino IDE upload command.

Hopefully, this workaround can help others until the upload command issue is resolved. ;)

Jason2866, 2023-12-04:
I have no H2 to test. So there is high chance there are maybe other bugs which only occurs with a real device connected.

Jason2866, 2023-12-04:
Mhh, probably this line needs to be changed too https://github.com/Jason2866/platform-espressif32/blob/Arduino/IDF5/builder/main.py#L430

Can you change from `"--flash_freq", "${__get_board_f_flash(__env__)}",` to `"--flash_freq", "${__get_board_img_freq(__env__)}",`

Jason2866, 2023-12-05:
Thx, for testing will merge the changes in branch `Arduino/IDF5`

closed 2023-12-05 by Jason2866