Jason2866/platform-espressif32 #56: Customized serial Rx pin is not functioning properly.

createskyblue, 2024-01-14:
Hello everyone,

I've encountered an issue while using the Espressif32 platform in the Tasmota project. I noticed that this problem arises when I reference the latest version of the platform: [platform-espressif32.zip](https://github.com/tasmota/platform-espressif32/releases/download/2024.01.00/platform-espressif32.zip). During my attempt to resolve the issue, I initiated a discussion on the [issues page of the platformio/platform-espressif32 project](https://github.com/platformio/platform-espressif32/issues/1279), but later realized that this might not be the correct place.

@Jason2866 suggested that I submit the issue on the [issues page of the espressif/arduino-esp32 project](https://github.com/espressif/arduino-esp32/issues), but when I tried to reproduce the problem with the latest Arduino ESP32 releases, I was unsuccessful. Unfortunately, it seems like this issue is more related to Tasmota's [platform-espressif32 project](https://github.com/tasmota/platform-espressif32).

I would appreciate some guidance on whether it is suitable to initiate a discussion regarding this issue on the platform-espressif32 project page of Tasmota, as I am uncertain about the appropriateness of doing so here. Your insights would be valuable.

# Problem description

The latest version of the Tasmota branch.
`platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.00/platform-espressif32.zip`

"After customizing the serial Rx pin, data reception is not working properly. Below is the minimal code that reproduces the issue."

[Esp32C3_dev_pio_Serial_test.zip](https://github.com/platformio/platform-espressif32/files/13932174/Esp32C3_dev_pio_Serial_test.zip)

```
#include <Arduino.h>

void setup() {
    Serial.begin(921600, SERIAL_8N1, 5, 4); // Rx = 5, Tx = 4
    Serial.print("READY\n\r");

}

void loop() {
    if (Serial.available()) {
        Serial.write(Serial.read());
    }
}

```

```
[env:esp32-c3-devkitm-1]
; platform = espressif32
platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.00/platform-espressif32.zip
board = esp32-c3-devkitm-1
framework = arduino
; change microcontroller
board_build.mcu = esp32c3
board_build.flash_mode = dio
; change MCU frequency
board_build.f_cpu = 160000000L
upload_protocol = esptool

; build_flags =
;     -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
;     -D CONFIG_ARDUHAL_LOG_COLORS=1
;     -D CONFIG_LWIP_IP_FORWARD=1
;     -D CONFIG_LWIP_IPV4_NAPT=1
;     -D CONFIG_LWIP_L2_TO_L3_COPY=1

;     -D ARDUINO_USB_CDC_ON_BOOT=0
;     -D ARDUINO_USB_MODE=0
  

;     ; -D USE_UDP_LOG
;     -DUSE_ESP_IDF_LOG -DLOG_LOCAL_LEVEL=5 -DTAG="\"ARDUINO\"" -DCONFIG_LOG_COLORS
debug_server =
  ${platformio.packages_dir}/tool-openocd-esp32/bin/openocd
  -f
  ${platformio.packages_dir}/tool-openocd-esp32/share/openocd/scripts/board/esp32c3-builtin.cfg
  
build_type = release
monitor_filters = esp32_exception_decoder
debug_tool = esp-builtin
upload_port = COM20
monitor_port = COM20
board_build.partitions = default.csv
board_build.filesystem = littlefs
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
```

The purpose of this code is to print the data received from the serial port, but it doesn't work properly in this plaofmtio version.

createskyblue, 2024-01-14:
I temporarily fixed this issue by downgrading four files under ./cores/esp32. They are:

_esp32-hal-uart.c
esp32-hal-uart.h
HardwareSerial.cpp
HardwareSerial.h_

I replaced them with the following files:
[esp32-hal-uart.zip](https://github.com/Jason2866/platform-espressif32/files/13932911/esp32-hal-uart.zip)

Jason2866, 2024-01-14:
What are your reasons for wanting to use Tasmota Arduino Core 2.0.14 instead of the espressif Platformio official Core 2.0.14?
As you already noticed there is no difference in the code regarding serial.
Tasmota Arduino Core 2.0.14 claims no compability to anything official released.
Since we have no issue using serial with Tasmota, there will be no investigation.

createskyblue, 2024-01-14:
> What are your reasons for wanting to use Tasmota Arduino Core 2.0.14 instead of the espressif Platformio official Core 2.0.14? As you already noticed there is no difference in the Arduino code regarding serial. Tasmota Arduino Core 2.0.14 claims no compability to anything official released. Since we have no issue using serial with Tasmota, there will be no investigation.

I want to use the NAPT (Network Address and Port Translation) functionality, which is available in Tasmota Arduino Core 2.0.14. This feature is crucial for my use case, and that's why I prefer Tasmota over the official espressif Platformio Core 2.0.14. 

In any case, thank you for your response.

closed 2024-01-14 by createskyblue