Jason2866/platform-espressif32 #60: examples/espidf-ulp-riscv

newmansnj, 2024-02-09:
examples/espidf-ulp-riscv
When I try to run your example on Windwos with framework = arduino, espidf I get the error message "The command line is too long" on linux the error message does not appear with the exact same project. The same error message appears when I run framework espidf (without arduino ) but with platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.12/platform-espressif32.zip or platform = espressif32 only with an exact version number @ ^ it does not appear because then the name for the version is smaller, no long strings follow for the exact espressif32 version.  

The reason is the cmd which is executed in windows and has a character limit of 8191.

Jason2866, 2024-02-09:
Don't know how i should fix a Windows issue.

Jason2866, 2024-02-09:
Maybe this? https://www.autodesk.com/support/technical/article/caas/sfdcarticles/sfdcarticles/The-Windows-10-default-path-length-limitation-MAX-PATH-is-256-characters.html

newmansnj, 2024-02-09:
No, the problem is the platform an outdated API uses what is still from DO's times, which limits the characters length to 8191.

Jason2866, 2024-02-09:
Closing since the issue is not in the fork of platform espressif32

closed 2024-02-09 by Jason2866