Jason2866/platform-espressif32 #67: missing sdkconfig.h

svefro, 2024-04-09:
After switching from:

````
board = esp32doit-devkit-v1
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.4.1/platform-espressif32-solo1-2.0.4.1.zip
````

to 
````
board = esp32-solo1
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
````

I get a error. "**sdkconfig.h: No such file or directory**"
i have tried to delete the ".platformio" folder and let platformio generate a new one. but no luck.
i need this to be able to compile my firmware for shelly pluss 1 pm (esp32 solo)

**How can i resolve this?**

**From compile output:**
````
C:/Users/svefr/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/assert.h:20:10: fatal error: sdkconfig.h: No such file or directory

*******************************************************************
* Looking for sdkconfig.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:sdkconfig.h"
* Web  > https://registry.platformio.org/search?q=header:sdkconfig.h
*
*******************************************************************

 #include <sdkconfig.h>
          ^~~~~~~~~~~~~
compilation terminated.
In file included from c:\users\svefr\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\sys-include\sys\reent.h:503,
                 from C:/Users/svefr/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\svefr\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\sys-include\stdio.h:60,
                 from C:/Users/svefr/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:27,
                 from src/library/Time.cpp:31:
````

Jason2866, 2024-04-09:
There is still something wrong with your setup. With
```
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
```
the toolchain `toolchain-xtensa-esp32@8.4.0+2021r2-patch5` is NOT used

closed 2024-04-09 by Jason2866

svefro, 2024-04-09:
c++ sure is something else... I have been away from it for a while.

It turned out it was because in my HTTPUpdateServer i had:
``#include <SPIFFS.h>``
used here:
````

 if (upload.name == "filesystem") {
     if (!Update.begin(SPIFFS.totalBytes(), U_SPIFFS)) {//start with max available size
         if (_serial_output) Update.printError(Serial);
     }
 }
 else {
    uint32_t maxSketchSpace = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
    if (!Update.begin(maxSketchSpace, U_FLASH)) {//start with max available size
        _setUpdaterError();
    }
````

SPIFFS.h was not found when i commented out "board_build.flash_mode = dout" in platformio.ini during debugging

Commented out the spiffs section and removed "board_build.flash_mode = dout" and it builds.

Jason2866, 2024-04-10:
As designed. Removed SPIFFS. It is deprecated. We use LittleFS.