Jason2866/platform-espressif32 #71: board_build.embed_txtfiles compilation error

rvbc1, 2024-06-04:
Hi everyone,

I'm currently using the following configuration:

```
platform = https://github.com/Jason2866/platform-espressif32.git#a16e5eac65ad2b2d1421de5c0bd9769365dbad74

board_build.embed_txtfiles = data/data.txt
```

However, I'm encountering a compilation error:

```
Building in release mode
prepare_file(["build/local/data.txt.txt.o"], ["data/data.txt"])
Converting build/local/data.txt.txt.o
riscv32-esp-elf-objcopy: architecture xtensa unknown
*** [build/local/data.txt.txt.o] Error 1
```

Interestingly, this issue does not occur in other PlatformIO projects using the default ESP32 platform.

I'd appreciate any help in resolving this compilation error.

Thanks!

Additional notes:

- This issue only occurs when using the custom ESP32 platform mentioned above.
- The default ESP32 platform works as expected.

Jason2866, 2024-06-04:
Without a full example setup, it is guess work. Please provide. Best push it in a repo.

rvbc1, 2024-06-05:
Sorry for the incomplete setup. I checked the last commit (6cd40876e4d47bd72996e2f3cedc1a1abe44ddbf) and I see the issue is solved. Thanks!

rvbc1, 2024-06-05:
For your information, the error occurred on the ESP32-H2 board

closed 2024-06-05 by Jason2866