Jason2866/platform-espressif32 #73: Arduino 3.x - stuck on boot with Wokwi.

PBrunot, 2024-07-03:
I have a platformio Arduino project which works fine with platform-espressif32 tagged 2024.04.01. Your platform distribution is awesom provides me +300kB memory savings versus the standard pio espressif32 !

However I have an issue: the same project, with platform tagged 2024.07.20 is always stuck on boot in the esp_hal_read function (this line on esp32s3 https://github.com/espressif/esp-idf/blob/e7070e777a079695f69720ffb3c631c5fe620cc6/components/hal/esp32s3/efuse_hal.c#L67 or this line on esp32s2 https://github.com/espressif/esp-idf/blob/e7070e777a079695f69720ffb3c631c5fe620cc6/components/hal/esp32s2/efuse_hal.c#L90 )

With GDB if I force the register to exit the infinite loop it gets stuck further down the line in other efuse functions.

This was observed with Wokwi extension and Wokwi online platform. 

I've seen this issue with Arduino-3.0.0-rc2, and now with Arduino-3.0.2 based release.
The issue was not present with the Arduino-3.0.0-rc3 release (2024.04.01).

This may be a Wokwi bug, but I'm not sure how to investigate further. I'm opening this issue if someone has the same issue and can provide troubleshooting steps.

Jason2866, 2024-07-04:
Release 2024.07.20 is special since it is based on IDF 5.2.2 and I **do** expect compability issues. The official Arduino release is based on IDF 5.1.4.
There is an actual release from me too. Released here https://github.com/tasmota/platform-espressif32/releases

PBrunot, 2024-07-04:
Thanks for your kind answer. 
I have the same exact behaviour with https://github.com/tasmota/platform-espressif32.git#2024.06.11 (Tasmota Arduino Core 3.0.2.240625 based on IDF v5.1.4.240625 ) - hang on boot with esp32s2 or esp32s3 (not tested with outhers)
Considering I'm stuck in early in the boot process on esp_efuse functions I suspect this issue is more compatibility issue between wokwi (online & VS extension) and ESP IDF.
It seems the Tasmota project is using wokwi too, is it working fine for you with the above release?

Jason2866, 2024-07-04:
No, we do not use wokwi with Tasmota. We tried a while ago and wokwi did get stuck early.
Tasmota has some very low level stuff in code and do linker wraps. Furthermore Platformio is forked and changed too. There is no part which is not altered.

Jason2866, 2024-07-05:
Just tested [this example](https://github.com/wokwi/esp32-ntp-clock) with my platform. Does work.
Used this platformio.ini
```
[env:esp32]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/2024.07.20/platform-espressif32.zip
framework = arduino
board = esp32-s3-devkitm-1
lib_deps = marcoschwartz/LiquidCrystal_I2C@^1.1.4
```

closed 2024-07-05 by Jason2866

PBrunot, 2024-07-07:
I have the same issue (hang on boot) with your example with the following configuration:

```text
[env:esp32]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/2024.07.20/platform-espressif32.zip
framework = arduino
board = esp32-s3-devkitm-1
lib_deps = marcoschwartz/LiquidCrystal_I2C@^1.1.4
```

* diagram.json

```json
{
  "version": 1,
  "author": "Uri Shaked",
  "editor": "wokwi",
  "parts": [
    { "type": "board-esp32-s3-devkitc-1", "id": "esp", "top": -78.19, "left": 44.14, "attrs": {} },
    {
      "type": "wokwi-lcd1602",
      "id": "lcd1",
      "top": -65.71,
      "left": 191.17,
      "attrs": { "pins": "i2c" }
    }
  ],
  "connections": [
    [ "esp:TX", "$serialMonitor:RX", "", [] ],
    [ "esp:RX", "$serialMonitor:TX", "", [] ],
    [ "lcd1:SDA", "esp:8", "green", [ "h-49.47", "v150.31" ] ],
    [ "lcd1:GND", "esp:GND.1", "black", [ "h-16.67", "v162.78" ] ],
    [ "lcd1:VCC", "esp:3V3.1", "red", [ "h-8.13", "v196.73" ] ],
    [ "lcd1:SCL", "esp:9", "purple", [ "h-27.75", "v-44.32" ] ]
  ]
}

```

* wokwi terminal output

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x370
load:0x403c9700,len:0x900
load:0x403cc700,len:0x2364
entry 0x403c98ac
```

* Verbose build extract

```text
Processing esp32 (platform: https://github.com/Jason2866/platform-espressif32/releases/download/2024.07.20/platform-espressif32.zip; framework: arduino; board: esp32-s3-devkitm-1; lib_deps: marcoschwartz/LiquidCrystal_I2C@^1.1.4)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitm-1.html
PLATFORM: Espressif 32 (2024.7.20) (https://github.com/Jason2866/platform-espressif32/releases/download/2024.07.20/platform-espressif32.zip) > Espressif ESP32-S3-DevKitM-1
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.2+sha.01c738d (https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/2567/framework-arduinoespressif32-all-release_v5.2-01c738d.zip) 
 - tool-esptoolpy @ 4.7.5 (https://github.com/Jason2866/esptool/releases/download/v4.7.5/esptool.zip) 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208 
 - toolchain-riscv32-esp @ 13.2.0+20230928 
 - toolchain-xtensa-esp-elf @ 13.2.0+20230928
[... 41 lines cut ...]
xtensa-esp32s3-elf-g++ -o "D:\GitHub\esp32-ntp-clock\src\esp32-ntp-clock.ino.cpp" -x c++ -fpreprocessed -dD -E "C:\Users\pasca\AppData\Local\Temp\tmp22x38uib"
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LiquidCrystal_I2C @ 1.1.4 (License: Unknown, Path: D:\GitHub\esp32-ntp-clock\.pio\libdeps\esp32\LiquidCrystal_I2C)
|   |-- Wire @ 2.0.0 (License: Unknown, Path: C:\Users\pasca\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
|-- WiFi @ 2.0.0 (License: Unknown, Path: C:\Users\pasca\.platformio\packages\framework-arduinoespressif32\libraries\WiFi)
|-- Wire @ 2.0.0 (License: Unknown, Path: C:\Users\pasca\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
Building in release mode
```

* I tried forcing xtensa toolchain to 12.2.0 from 13.2.0 with your latest release, but it still fails.

```
[env:esp32]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/2024.07.20/platform-espressif32.zip
platform_packages = toolchain-xtensa-esp32s3 @ 12.2.0+20230208
framework = arduino
board = esp32-s3-devkitm-1
lib_deps = marcoschwartz/LiquidCrystal_I2C@^1.1.4
```

* I'm running VSCode con amd64 platform, windows 11.

Jason2866, 2024-07-07:
Looks like a Platformio issue. You can try to delete the hidden folder `.platformio` with all it's content. As said before can't reproduce the issue.

PBrunot, 2024-07-07:
* Tried deleting .platformio, .pio - same issue
* Tried generating the BIN file from Ubuntu machine - same issue

So I've forked your example to reproduce the issue: https://github.com/PBrunot/esp32-ntp-clock/tree/esp32s3-test
I've run a Github action which build and generate a BIN file  : https://github.com/PBrunot/esp32-ntp-clock/actions/runs/9826190829
The BIN file hangs on boot (https://wokwi.com/projects/361252421391334401 and wokwi-cli)

I think this excludes my machine configuration. 

Can you share your working esp32s3 project with https://github.com/Jason2866/platform-espressif32/releases/download/2024.07.20/platform-espressif32.zip? I'd like to review any differences with my forked project.

Jason2866, 2024-07-07:
It is exactly as i wrote here https://github.com/Jason2866/platform-espressif32/issues/73#issuecomment-2210797122 with the described change of the Platformio.ini
Maybe it is platform dependent. Don't use Windows. Use Apple MacOS with M1 CPU

PBrunot, 2024-07-09:
FYI in case more people come into this, this issue was reproduced and it turned out to be a bug in Wokwi for efuse emulation. 
See https://github.com/wokwi/wokwi-features/issues/840