Jason2866/platform-espressif32 #81: C2 fails to compile on FastLED with this repo

zackees, 2024-09-19:
[image]

Let me know if I'm doing anything wrong. Here is my board defs for this product:

```python
ESP32_C2_DEVKITM_1 = Board(
    board_name="esp32-c2-devkitm-1",
    platform_needs_install=True,  # Install platform package to get the boards
    platform="https://github.com/Jason2866/platform-espressif32",  # No support from PIO, have to use a hacked version to get this to work
)
```

Jason2866, 2024-09-19:
CI example compiles fine with the C2
https://github.com/Jason2866/platform-espressif32/blob/fc0da1b9edc2548cbe882b65e6e3df794d1c51d2/examples/arduino-blink/platformio.ini#L23
https://github.com/Jason2866/platform-espressif32/actions/runs/10937501644/job/30363510554

Don't know what you show. Not a usual Platformio setup.

Jason2866, 2024-09-19:
Don't know the context but this is not a valid platform entry
```
platform="https://github.com/Jason2866/platform-espressif32
```
use
```
platform="https://github.com/Jason2866/platform-espressif32.git
```

Jason2866, 2024-09-19:
Closing since not reproducable. C2 compiles successfully and flashing does work as expected

closed 2024-09-19 by Jason2866

zackees, 2024-09-19:
So the issue here is that `pio ci` won't run, but `pio run` will.

I was able to create a hack that made it work for this board but holy crap.