Jason2866/platform-espressif32 #93: tasmota / arduino-esp32 issue with mDNS lib

vortigont, 2024-10-31:
Hi @Jason2866,
can't open the issue in [tasmota/arduino-esp32](https://github.com/tasmota/arduino-esp32), so created it here.
Found  a problem in recent tasmota's arduino-esp32 release [3.1.0.241030](https://github.com/tasmota/arduino-esp32/releases/tag/v3.1.0.241030).
For mDNS lib - the linker fails to build a project if mDNS header `ESPmDNS.h` is included somewhere in the project.
I found that file `libraries/ESPmDNS/src/ESPmDNS.cpp` supplied in the package is modified against the upstream source,
this line is added at the top
`#if CONFIG_MDNS_MAX_INTERFACES`
Looks like when the cpp unit is built it does not include `"sdkconfig.h"` so the `#if` guard is always `false`.
I fixed it with
```
#include "sdkconfig.h"
#if CONFIG_MDNS_MAX_INTERFACES
```

Could you pls check if this?
Appreciate your work on IDF's 5.3 build, so nice to have a new version of c++ toolchain here!
Thanks!

Jason2866, 2024-10-31:
Hi Vortigont,
Ahh, yes. Fixed. Use ```platform = https://github.com/tasmota/platform-espressif32/releases/download/2024.11.30/platform-espressif32.zip```

vortigont, 2024-10-31:
got it! Thanks so much!

vortigont, 2024-11-07:
Still not fixed, unfortunately :(
Inside this zip `https://github.com/tasmota/platform-espressif32/releases/download/2024.11.30/platform-espressif32.zip`
the manifest declares
```
  "packages": {                                                                                                                                               
    "framework-arduinoespressif32": {                                                                                                                         
      "type": "framework",                                                                                                                                    
      "optional": true,                                                                                                                                       
      "owner": "tasmota",                                                                                                                                     
      "version": "https://github.com/tasmota/arduino-esp32/releases/download/v3.1.0.241030/framework-arduinoespressif32.zip"                                  
    },   
```

and inside `https://github.com/tasmota/arduino-esp32/releases/download/v3.1.0.241030/framework-arduinoespressif32.zip`
the file `libraries/ESPmDNS/src/ESPmDNS.cpp` is still broken with 

```
#if CONFIG_MDNS_MAX_INTERFACES
```

Jason2866, 2024-11-07:
@vortigont Mhh why?
```
#include "sdkconfig.h"
#if CONFIG_MDNS_MAX_INTERFACES
.
.
.
```
So it should work

vortigont, 2024-11-07:
ah... so sorry... looks like it stuck in CI's cache. Indeed when downloaded with wget it's all fine.
Now builds fine!
Sorry for the noise!

closed 2024-11-07 by vortigont