Jason2866/platform-espressif32 #99: WifiSecure

Ifousoft, 2024-11-04:
Thanks for this libs that helps me to have esp32s3 working with 7 inches display and Wifi on pio.
What about WifiSecure that I can't find there ?

Thanks!

Jason2866, 2024-11-04:
It is removed.

Jason2866, 2024-11-04:
A full compatible version is here https://github.com/pioarduino/platform-espressif32

closed 2024-11-04 by Jason2866

Ifousoft, 2024-11-07:
Thanks. Nevertheless I have a problem with my 7 inches screen on ESP32S3.
With the platform propose by the seller, using Wifi create a scroll on the screen each few seconds...
Using the version your provided in the fork solves this problem.

But, with the new version you propose, I got the WiFisecure... but it restarts to scroll...

Damned!

Jason2866, 2024-11-07:
Probably the change sdkconfig how I compile the S3 OPI PSRAM Arduino variants libs
https://github.com/Jason2866/esp32-arduino-lib-builder/blob/v5.3_Arduino/configs/defconfig.opi_ram

Ifousoft, 2024-11-09:
OK... But could you please document a little more ? I don't know what to do with your comment...
Thanks!

Jason2866, 2024-11-09:
The changes affects the Arduino libs compile. The settings optimize performance for use with Displays

Ifousoft, 2024-11-09:
But, How could I use this information to solve my issue on the fact that the display is scrolling... alone... :-(

Jason2866, 2024-11-09:
Compiling the Arduino libs with this changed settings.

Ifousoft, 2024-11-14:
Could you explain me how to recompile arduino libs ? I have no access to sdkconfig because using platformio under arduino framework.
I found in blogs that pio -t menuconfig works only fr esp-idf framework and not Arduino one.
Morevover I identified that " If you are using Arduino as the framework, it probably won't work. Arduino comes with a precompiled sdkconfig file. You can try making sure you have selected to use the esp-idf framework."...

Sorry...

Ifousoft, 2024-11-14:
Could it be possible that you recompile a specific arduino framework with the requested config and put on github rather than 
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.07/platform-espressif32.zip

? Thanks!