Jason2866/platform-espressif32 #105: Do exceptions work?

Carlos4621, 2024-11-15:
I am trying to make a program with exception handling but when an exception is thrown abort() is automatically called, I use the PlatformIO addon in Visual Code. I made a simple code to test them, I leave you the code, the .ini file and the .cpp

platformio.ini

```platformio.ini
[env:heltec_wifi_lora_32_V3]
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
board = heltec_wifi_lora_32_V3
framework = arduino
build_flags = -fexceptions
build_src_flags = 
	-fexceptions
```

main.cpp

```main.cpp
#include <Arduino.h>
#include <stdexcept>

void setup() {
  Serial.begin(9600);
}

void loop() {
  try {
    throw std::runtime_error{ "exception!" };
  }
  catch(const std::exception& e) {
    Serial.println(e.what());
  }
  
}

```

Output:

```output.sh
abort() was called at PC 0x42008a52 on core 1

Backtrace: 0x4037621a:0x3fcec0c0 0x4037950d:0x3fcec0e0 0x4037c632:0x3fcec100 0x42008a52:0x3fcec170 0x4200ca55:0x3fcec190 0x42000ec7:0x3fcec1b0 0x42002550:0x3fcec1d0

ELF file SHA256: 53a1b3f1a24d891f
```

Jason2866, 2024-11-15:
The are disabled in IDF to save code space.

closed 2024-11-15 by Jason2866

Jason2866, 2024-11-15:
No, when removed it is gone. The full blown set is here https://github.com/pioarduino/platform-espressif32