Jason2866/platform-espressif32 #110: 'GPIO' was not declared in this scope

oo-dev17, 2024-12-29:
Hi!
Any idea why I am getting  'GPIO' was not declared in this scope during compilation of the OneWire library?

platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53
board = esp32-c6-devkitc-1
framework = arduino

Thanks in advance!

<details>
  <summary>Detailed log</summary>

Processing fbhMonitor (platform: https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53; board: esp32-c6-devkitc-1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitc-1.html
PLATFORM: Espressif 32 (2024.11.31+sha.5e94744) > Espressif ESP32-C6-DevKitC-1
HARDWARE: ESP32C6 160MHz, 320KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.1.0+sha.aa410416 
 - tool-esptoolpy @ 4.8.5 
 - tool-mklittlefs @ 3.2.0 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-riscv32-esp @ 13.2.0+20240530
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- DallasTemperature @ 3.11.0
|-- NTPClient @ 3.2.1
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- OneWire @ 2.3.8
|-- ArduinoOTA @ 3.1.0
|-- ESPmDNS @ 3.1.0
|-- Ticker @ 3.1.0
|-- WebServer @ 3.1.0
|-- WiFi @ 3.1.0
Building in release mode
Using manually specified: ttyACM0
Compiling .pio/build/fbhMonitor/src/main.cpp.o
Building .pio/build/fbhMonitor/bootloader.bin
Generating partitions .pio/build/fbhMonitor/partitions.bin
esptool.py v4.8.5
Creating esp32c6 image...
Merged 2 ELF sections
Successfully created esp32c6 image.
Compiling .pio/build/fbhMonitor/libe32/OneWire/OneWire.cpp.o
Compiling .pio/build/fbhMonitor/lib0b1/DallasTemperature/DallasTemperature.cpp.o
.pio/libdeps/fbhMonitor/OneWire/OneWire.cpp:599:22: warning: extra tokens at end of #undef directive
  599 | #  undef noInterrupts() {portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;portENTER_CRITICAL(&mux)
      |                      ^
.pio/libdeps/fbhMonitor/OneWire/OneWire.cpp:600:20: warning: extra tokens at end of #undef directive
  600 | #  undef interrupts() portEXIT_CRITICAL(&mux);}
      |                    ^
Compiling .pio/build/fbhMonitor/lib915/NTPClient/NTPClient.cpp.o
In file included from .pio/libdeps/fbhMonitor/OneWire/OneWire.cpp:149:
.pio/libdeps/fbhMonitor/OneWire/util/OneWire_direct_gpio.h: In function 'uint32_t directRead(uint32_t)':
.pio/libdeps/fbhMonitor/OneWire/util/OneWire_direct_gpio.h:173:17: error: 'GPIO' was not declared in this scope
  173 |         return (GPIO.in >> pin) & 0x1;

</details>

Jason2866, 2025-01-05:
Use actual master branch of the onewire lib

closed 2025-01-05 by Jason2866

oo-dev17, 2025-01-05:
Got it, thanks a lot Jason