Jason2866/platform-espressif32 #135: Pio fails to configure when opening project after working on another project that uses different espressif32 platform.

yigitabi5444, 2025-05-14:
Pio crashes when opening project after working on another project that uses a different espressif32 platform 

Steps to replicate on afresh platformio install:
- Open project 1, let it configure and download tooling.
- Open project 2, let it configure and download tooling.
- Try opening project 1 again, it will fail to configure and throw an error

Only solution I found so far is deleting .platformio and letting everything reinstall. Which I frequently need to do when switching between projects.

"Project 1"'s ini configuration:
  [env:esp32-c3-devkitm-1]
  platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53
  board = esp32-c3-devkitm-1
  framework = arduino, espidf

"Project 2"'s configuration:
  [env:esp32-s3-devkitc-1]
  platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
  board = esp32-s3-devkitc-1
  framework = arduino

Crash log:

Resolving esp32-c3-devkitm-1 dependencies...
Wrong version for toolchain-riscv32-esp. Installing needed version...
Tool Manager: Installing https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/3616/framework-arduinoespressif32-all-release_v5.3-88cbd669.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32@3.1.3+sha.88cbd669 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/riscv32-esp-elf-13.2.0_20240530.zip
Unpacking 0% 10% 20%
Tool Manager: toolchain-riscv32-esp@13.2.0+20240530 has been installed!
Tool Manager: Installing https://github.com/tasmota/esptool/releases/download/v4.8.10/esptool.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esptoolpy@4.8.10 has been installed!
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing esp32-c3-devkitm-1 (platform: https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF53; board: esp32-c3-devkitm-1; framework: arduino, espidf)
--------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/riscv32-esp-elf-13.2.0_20240530.zip
Unpacking 0% 10% 20%
Tool Manager: toolchain-riscv32-esp@13.2.0+20240530 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (2025.5.30+sha.019ad93) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.1.3+sha.88cbd669 
 - framework-espidf @ 3.50303.0 (5.3.3) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 4.8.10 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.10.2 
 - tool-scons @ 4.40801.0 (4.8.1)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType:
  File "/Users/yigit/.platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/yigit/.platformio/platforms/espressif32@src-979d69a3a2d3d7b03ac8a6f8f700424c/builder/main.py", line 370:
    target_elf = env.BuildProgram()
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/Users/yigit/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/Users/yigit/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 142:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Util/envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "/Users/yigit/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 352:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 684:
    return method(*args, **kw)
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/yigit/.platformio/packages/tool-scons/scons-local-4.8.1/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/yigit/.platformio/platforms/espressif32@src-979d69a3a2d3d7b03ac8a6f8f700424c/builder/frameworks/espidf.py", line 148:
    assert os.path.isdir(TOOLCHAIN_DIR)
  File "<frozen genericpath>", line 42:
    
========================= [FAILED] Took 21.29 seconds =========================

Jason2866, 2025-05-14:
It is a migration issue. Since a few commits Platformio Registry is not used anymore. Using a slightly modified installer from espressif IDF. All tools and toolchains are loaded from Open Source Repos release section.

To solve:
- close VSC
- search in user folder the hidden folder `.platformio`
- change from `.platformio` to folder `packages`
- delete any files and folders but not the folder `pyenv`
- open VSC
- start a compile run

Jason2866, 2025-07-06:
fixed

closed 2025-07-06 by Jason2866