Jason2866/platform-espressif32 #146: C2 compiles, but does not run

hb020, 2025-05-30:
The below is a pure copy of https://github.com/platformio/platform-espressif32/issues/1151#issuecomment-2921970097

But since you said "Do NOT open an issue here when you encounter problems." and since the problem is in both the platformio and this repo, I just crosspost. I know, should not, but just trying to do as I'm told.

-----
Have gotten nowhere so far.

Device: ESP8684-WROOM-04C with 2MB flash (so not in the devkit, nor a ESP8684-MINI-1, but that shouldn't change much. Maybe I'm overlooking something, but apart from the pinout and the antenna, the schemas look identical)

Strapped EN, IO8 and IO9 to high. Buttons on EN and IO9 for resp. reset and flash mode.

```
[platformio]
default_envs = esp32-c2-devkitm-1

[env:esp32-c2-devkitm-1]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino
board = esp32-c2-devkitm-1
monitor_speed = 115200
build_flags =
custom_component_remove = espressif/esp_hosted
                          espressif/esp_wifi_remote
                          ....... 
```
.. so essentially the arduino_blink example.

No matter what platform I use:
* `platform = https://github.com/Jason2866/platform-espressif32.git`
* `platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip`
* `platform = https://github.com/pioarduino/platform-espressif32.git`

I can compile and upload, but all programs I uploaded result in the same: 
* no output on serial, not even glitches during boot
* no control over the pins, they are all static (either GND or VCC), not even glitches during boot, except IO6, which has a glitch to GND about once every 3 seconds

Of course I wiped all `~/.platformio/platforms/espressif32` directories and `~/.platformio/packages` between platform changes.

Example of simple program I tested:
```
#include <Arduino.h>
void setup()
{
  Serial.begin(115200);
  Serial.println("Setup done");
}

void loop()
{
  delay(1000);
  Serial.println("Loop");
}
```

What are the settings required? 

FYI, the upload uses the following parameters to esptool:

```
--chip esp32c2 
--port "/dev/cu.usbserial-FTHJGC2Z" 
--baud 460800 
--before default_reset 
--after hard_reset 
write_flash 
-z 
--flash_mode dio 
--flash_freq 60m 
--flash_size detect 
0x0000 .pio/build/esp32-c2-devkitm-1/bootloader.bin 
0x8000 .pio/build/esp32-c2-devkitm-1/partitions.bin 
0xe000 ~/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 
0x10000 .pio/build/esp32-c2-devkitm-1/firmware.bin
```

esptool produces the following:

 > General output:
esptool.py v4.8.9
Serial port /dev/cu.usbserial-FTHJGC2Z
Connecting....
Chip is ESP32-C2 (revision v1.2)
Features: WiFi, BLE
Crystal is 26MHz
MAC: c0:5d:89:50:f8:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.

> chip_id
Warning: ESP32-C2 has no Chip ID. Reading MAC instead.
MAC: c0:5d:89:50:f8:78

> flash_id 
Manufacturer: a1
Device: 4015
Detected flash size: 2MB

> read_flash_status
Status value: 0x0200

> get_security_info
Security Information:
=====================
Flags: 0x00000000 (0b0)
Chip ID: 12
API Version: 1
Secure Boot: Disabled
Flash Encryption: Enabled
SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x7

Using `--flash_mode dio` or `--flash_mode qio` makes no difference.

Jason2866, 2025-05-30:
since you have 2MB of flash you need a matching setup. You can use this boards manifest https://github.com/Jason2866/Tasmota-build/blob/special_builder/boards/esp32c2_2M.json

closed 2025-05-30 by Jason2866

Jason2866, 2025-05-31:
Can not reproduce my two C2 both works fine. You can install Tasmota from here via WebInstaller to see if you module is defect.
https://jason2866.github.io/Tasmota-specials/

hb020, 2025-05-31:
It appears my devices are crippled. I try to reprogram the riden dongles that have started using ESP8684 since not too long ago. From what I understand, they have protected the devices against reflashing via the efuses: SPI_BOOT_CRYPT_CNT = 0b111, meaning: _Release mode_.