Jason2866/platform-espressif32 #162: Pio fails to configure on latest IDF53 branch

yigitabi5444, 2025-07-01:
# platformio.ini config:

```
[platformio]
default_envs = esp32-s3-devkitc-1
core_dir = .platformio

[env:esp32-s3-devkitc-1]
platform = https://github.com/Jason2866/platform-espressif32#Arduino/IDF53
board = esp32-s3-devkitc-1
framework = arduino, espidf
board_build.arduino.memory_type = qio_qspi
board_build.flash_mode = qio
board_build.psram_type = qio
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
board_build.extra_flags = 
  -DBOARD_HAS_PSRAM
board_build.partitions = partitions.csv
```

# Steps to recreate the issue:
- Delete .platformio and .pio folders
- Restart vscode, let it configure
- following error pops up, looks like idf_tools.py doesnt install properly somehow.

Unrelated question, is there a stable branch you recommend I follow to get idf 5 and arduino 3 frameworks working together in platformio?

```
********************************************************************************
If you like PlatformIO, please:
- star it on GitHub > https://github.com/platformio/platformio-core
- follow us on LinkedIn to stay up-to-date on the latest project news > https://www.linkedin.com/company/platformio/
********************************************************************************

Resolving esp32-s3-devkitc-1 dependencies...
Platform Manager: Installing git+https://github.com/Jason2866/platform-espressif32#Arduino/IDF53
git version 2.39.5 (Apple Git-154)
Cloning into '/Users/aaronaders/Documents/GitHub/fw/.platformio/.cache/tmp/pkg-installing-o9eunnze'...

Platform Manager: espressif32@2025.6.30+sha.c915f32 has been installed!
Tool Manager: Installing https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/3758/framework-arduinoespressif32-all-release_v5.3-ea465b8d.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32@3.1.3+sha.ea465b8d has been installed!
Tool Manager: Installing https://github.com/tasmota/esp-idf/releases/download/v5.3.3.250618/esp-idf-v5.3.3.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-espidf@3.50303.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-13.2.0_20240530.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: toolchain-xtensa-esp-elf@13.2.0+20240530 has been installed!
Tool Manager: Installing https://github.com/Jason2866/esptool/releases/download/v5.0.0-dev1/esptool.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esptoolpy@5.0.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp_install/releases/download/v5.0.0/esp_install-v5.0.0.zip
Downloading 0% 10% 20% 30% 40% 50% 60%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tl-install@5.0.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/cmake-3.30.2.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: tool-cmake@3.30.2 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esp-rom-elfs-20241011.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/ninja-1.10.2.zip
[... 101 lines cut ...]
    xtensa-esp32s3-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

CMake Error at .platformio/packages/framework-espidf/tools/cmake/project.cmake:571 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32s3-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 21.98 seconds =========================
```

Jason2866, 2025-07-02:
Use the Tasmota platform releases. There the setup is fix. All Tasmota releases are based on this repo

closed 2025-07-02 by Jason2866

psamoilov, 2025-07-24:
@Jason2866 I face this as well at tasmota. 2025.04.30 still works, later releases doesn't work. Wiping .platformio, reinstall etc doesn't help. 

Processing v2 (platform: https://github.com/tasmota/platform-espressif32/releases/download/2025.07.31/platform-espressif32.zip; board: esp32-c3-devkitm-1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
Verbose mode can be enabled via `-v, --verbose` option
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
error: /Users/x/.platformio/packages/tool-esptoolpy does not appear to be a Python project, as neither `pyproject.toml` nor `setup.py` are present in the directory
Warning: Failed to install esptool: Command '['/Users/x/.platformio/penv/bin/uv', 'pip', 'install', '--quiet', '--python=/Users/x/.platformio/penv/bin/python', '-e', '/Users/x/.platformio/packages/tool-esptoolpy']' returned non-zero exit status 2.
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c3-devkitm-1.html
PLATFORM: Espressif 32 (2025.7.31) > Espressif ESP32-C3-DevKitM-1
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - contrib-piohome @ 3.4.4 
 - framework-arduinoespressif32 @ 3.1.3+sha.a0094062 
 - tool-esptoolpy @ 5.0.1 
 - tool-mklittlefs @ 3.2.0 
 - toolchain-riscv32-esp @ 13.2.0+20240530
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ strict