Jason2866/platform-espressif32 #208: PSRAM issue with custom_sdkconfig (ESP32 S3N8)

psamoilov, 2025-09-20:
If I use custom_sdkconfig directive in platformio.ini, e.g.

`custom_sdkconfig = '# CONFIG_LWIP_IPV6 is not set'
	'# CONFIG_LWIP_IPV6_AUTOCONFIG is not set'`

PSRAM is not initialized correctly, so code

`Serial.printf("Total heap: %d\n", ESP.getHeapSize());
Serial.printf("Free heap: %d\n", ESP.getFreeHeap());
Serial.printf("Total PSRAM: %d\n", ESP.getPsramSize());
Serial.printf("Free PSRAM: %d\n", ESP.getFreePsram());`

displays 

`Total heap: 298692
Free heap: 257912
Total PSRAM: 0
Free PSRAM: 0
`

without custom_sdkconfig everything works fine

`Total heap: 293460
Free heap: 253072
Total PSRAM: 7143424
Free PSRAM: 7140692 `

Jason2866, 2025-09-20:
This is the PSRAM hell from the esp32-s3. You have to set the correct and needed PSRAM settings for your board.
Arduino does compile all variants for the S3. The matching one is choosen depending on the boards.json.
HybridCompile is a variant of compile Arduino as an component of IDF. In this build variant most infos from the boards json are **not** used. This is the exact same behavior as it always was / is in Platformio.

I may add this feature in future.

Jason2866, 2025-10-01:
You can try branch https://github.com/Jason2866/platform-espressif32/tree/IDF55_Hybrid_PSRAM
use in platformio.ini with:

```
https://github.com/Jason2866/platform-espressif32.git#IDF55_Hybrid_PSRAM
```